kylecorry31/Trail-Sense #249: Unkonwn beacon pointer in compass

aw-bib, 2020-10-18:
Having reset the app some time ago I currently do not have any beacons defined. Funny enough the compass shows one beacon in light blue. I wonder where this one results from. Adding a new beacon doesn't change it. Heading towards it does not give any information.

kylecorry31, 2020-10-18:
The light blue pointer is actually a set heading, and it appears when you tap the compass. It is used if you want to set a heading but not have a location attached to it. I have an item to review the user experience of that feature, and make it clearer that it is a waypoint and you can tap to enable/disable

kylecorry31, 2020-10-18:
Closed in favor of #180

closed 2020-10-18 by kylecorry31

aw-bib, 2020-10-19:
> The light blue pointer is actually a set heading, and it appears when you tap the compass

Ah. Actually, a cool feature. Thanks :)