kylecorry31/Trail-Sense #267: No location on LineageOS 16

dotsumae, 2020-11-13:
Hello,
I have been using your app on two different versions of LineageOS. Your app works fine on LineageOS 17 GSI, but cannot display my location on LineageOS 16. I am stuck for a few months on LOS 16 : how can I help with this bug ?
I have no error message, only 0°0'0.0" N 0°0'0.0"E for a location, 0m for altitude, and unknown precision. Everything else seems to work.
I have checked that Trail Sense is allowed to access my location, and I disabled battery optimizations.

kylecorry31, 2020-11-13:
Hi,

Can you try a few things for me?

- Under Trail Sense's Settings click GPS and ensure auto adjust location is on and a location is visible at the top of the page
- Can you confirm that the GPS is working fine in other applications?

dotsumae, 2020-11-13:
[image]
Auto adjust is enabled, and my location is not visible at the top of the page.
Two of my other app can locate me just fine (Magicearth and Tisséo)

kylecorry31, 2020-11-13:
Thank you. 

(I'm not familiar with the settings on Lineage OS, so this might not be available)

Can you option your phone's settings and check the status of the following:
Settings > Security > Location

Trail Sense does not use network provided location, so the value needs to be Device Only, or High Accuracy (at least that's what it used to be called on AOSP)

Let me know if that helps.

It may also be useful to see what happens if you set it to device only and then open another app which uses the GPS.

dotsumae, 2020-11-13:
I checked Trail Sense's permissions, and it has "access precise location (GPS and network-based)" enabled. That sounds like "high accuracy".
In the notification menu, I have a notification switch that can be set on "High accuracy" (gps, wifi,bt,cell network), Battery saving (wifi,bt,cell) or "device only" (gps only)

In order to test more stuff, I downloaded and tested two more apps that have to use my location. Here is a summary of my tests : 

- On High accuracy, Geovelo, Tisséo, Magic Earth, and StreetComplete can locate me.
- On Battery saving, only Tisséo can locate me.
- On Device only, Geovelo and Tisséo can locate me.

Only trail sense fails to locate me. 

I should have mentioned, I am using LineageOS with microG. That means Google's location service is replaced by (UnifiedNlp)[https://github.com/microg/UnifiedNlp]. 
I will report those strange behaviours to LOS16's devs, as I had no issue with Trail sense on a (extra experimenal) LOS17 build. The issue might be on the OS side, or how your app deals with microG. If it can help, I am using a Fairphone 3.

kylecorry31, 2020-11-14:
Thank you for the detailed information. I'll try to explore some reasons why Trail Sense isn't getting location in this case - and may end up adding a screen within Trail Sense to help with identifying issues like this. 

Something to note is that Trail Sense does not use Google Play Services at all, and relies on the built-in Android location manager - so I wouldn't expect microG to be the problem.

dotsumae, 2020-11-26:
Hi !
My problem got fixed with the latest update (1.2.1). My OS did not get updated. Thank you for the update :) Thank you also for your app in general I love the idea and the design !
I still cannot create a beacon with the "USE CURRENT LOCATION" button, but this is another issue and not a crucial functionality so I'm fine using paper. A nice option would be to be able to copy my coordinates to clipboard.

closed 2020-11-26 by dotsumae

kylecorry31, 2020-11-29:
Glad to hear!

I would expect the USE CURRENT LOCATION button to work, as it uses the same GPS service as the location on the navigation tab. It should now display a loading indicator when it is trying to get a GPS fix. Does it work if you let it sit for a while? The timeout should be 8 seconds, but that's assuming GPS readings are coming in, just without a satellite count (or a low count). 

Might be worth adding a full timeout to the GPS, which will use your last known location (if available) when there is no GPS signal (which is what the navigation tab does).