kylecorry31/Trail-Sense #277: Acknowledge alerts

aw-bib, 2020-11-17:
If I get a sun set alert I would acknowledge it by swiping it away. However, Trail Sense keeps asking me if I want to continue these notifications all the time. Is this intended or am I doing something wrong here?

kylecorry31, 2020-11-17:
Hi, this seems to be an alert on your Android device itself - I would assume you might be able to disable that on your phone's settings. Unfortunately, there's nothing I can do from within Trail Sense to prevent this from happening.

closed 2020-11-17 by kylecorry31

aw-bib, 2021-01-01:
A happy new year! :)

Just for documentation, I finally understood the behaviour.

The alert is triggered => a notification goes off. Say: "Sun is setting at 16:01" fires at `15:01`. If I now try to acknowledge the notification by swiping it aside, Android asks me if I want to get further notifications from TrailSense and I could _switch it off_ there. But of course I want to get it tomorrow => `Keep showing`. This is the behaviour I mentioned in my question above.

If I happen to have _more patience_ say till `16:05` and _then_ acknowledge the notification it is removed right away, no asking and the alert will show up tomorrow again.

kylecorry31, 2021-01-01:
Thank you for the documentation - I'll add that to FAQ in case anyone else has the same question. Happy new year to you as well!