kylecorry31/Trail-Sense #292: Recommend Survival Manual and OsmAnd

notramo, 2020-11-20:
These two are great open source software, which can be a good companion to Trail Sense, but lots of users don't know about it.

Trail Sense should detect if they are installed, and recommend only those not.

kylecorry31, 2020-11-20:
I was thinking the same thing :)

For now, it would make sense for me to mention them on the README / User Guides. I can contact the developers to see if they would be okay with that integration - and also, from my work on Trail Sense Maps, I needed to modify its source code to support being detected by Trail Sense (if you were to install TS Maps - which isn't ready - you would see an 'Offline Maps' tool under the Tools tab)

notramo, 2020-11-21:
I was thinking about a dialog:
### **Recommended apps**
Get the most out of your phone when it's needed:
- Survival manual:
Offline guide for surviving
**Install**
- OsmAnd
Offline maps, track recording, navigation
**Install**

kylecorry31, 2020-11-21:
Rather than a dialog, I made this a guide.

closed 2020-11-21 by kylecorry31

notramo, 2020-11-21:
@kylecorry31 You should mention that F-Droid has the OsmAnd+ version for free, while it is paid in Google Play.

commit b80cca72 (2020-11-21): Add recommended apps guide

Closes #292