kylecorry31/Trail-Sense #310: Get Trail Sense "Play Verified"

kylecorry31, 2020-12-06:
Not exactly sure what the steps are to do so, need to do some research first.

qwerty287, 2021-04-11:
What does "Play Verified" mean? Does it mean "Verified by Play Protect"? I tihnk, there's no "Play verified", I only know this "Verified by Play Protect". If it'd mean "Verified by Play Protect", you can close the issue. (Trail Sense is already verified by Play Protect.)

kylecorry31, 2021-04-11:
Okay, thanks! Someone left that in a review on the Google Play store, and it showed as verified by play protect for me (seemed it did that by default, but I didn't have another device to check with) - I created this to look into if there was anything else I needed to do (as I didn't find anything with a quick google search).

closed 2021-04-11 by kylecorry31