kylecorry31/Trail-Sense #311: Add tape measure tool

kylecorry31, 2020-12-06:
(within ruler tool)

1. Open the ruler
2. Tap and hold the ruler (near the start)
3. Do not move finger and slide the phone under the finger - the ruler will slide, like a tape measure
4. Lift finger and repeat step 2 until complete

ChildishGiant, 2021-05-19:
This could also be implemented with AR. Google have such an implementation already: https://play.google.com/store/apps/details?id=com.google.tango.measure

kylecorry31, 2021-05-19:
That would be interesting to do, it would probably take a lot of work to get right - there are some reasons why Trail Sense can't use Google Play Services, which I believe that Google framework is in. I will definitely explore it though