kylecorry31/Trail-Sense #314: Canvas Map

kylecorry31, 2020-12-08:
A "map" which the user can draw features on to mark their surroundings. Possibly supporting the idea of tracks, which records the user's tracks and displays them (or just allowing the user to add a waypoint to the map). 

My only concern would be how to convey scale to the user, as showing scale (or zoom) is easy when you have actual map tiles, but without that, it may be hard for the user to draw features. A way I can think about getting around it is allowing them to place "features" along the path (rather than draw), which are very similar to beacons, but might not be something you want to navigate to / see on the navigation tab (but maybe have the option to within the "map").

bielej, 2020-12-21:
If possible - it could use previously downloaded maps from [osmand](https://github.com/osmandapp/OsmAnd).

kylecorry31, 2021-04-08:
Closed in favor of #516

closed 2021-04-08 by kylecorry31