kylecorry31/Trail-Sense #319: Get location without GPS

kylecorry31, 2020-12-12:
LONGITUDE

The time of solar noon needs to be known for longitude

LATITUDE

The angle of the sun needs to be known for latitude (may be hard to get without using the camera, as sighting the phone down its spine can't be used when looking directly at the sun)

OR

The length of the shadow of a known height object.

OR

The altitude of the north star (only works in the northern hemisphere)

kylecorry31, 2020-12-12:
**Longitude**

Given you have your latitude, the longitude can be calculated as the difference (in hour angle) between your local solar time and UTC. Local solar time can be calculated using this method: https://astronavigationdemystified.com/2013/05/08/the-survival-sundial-2/ (for which the compass can be used)

But if the time of solar noon is known (when the sun is either directly north or south), then the latitude does not need to be known to calculate longitude.

kylecorry31, 2020-12-12:
**Latitude**

Given you are in the northern hemisphere and it is night time, the altitude of the North Star (Polaris) is your latitude.

The angle of the sun can also be used - more research is needed on the calculation required for this. I might be able to brute force a calculation of the latitude using astronomy calculations and choosing the closest match for the current UTC time.

kylecorry31, 2020-12-15:
**Both**
If you know the altitude (technically 90 - altitude) of 3 well-known stars, you can get your latitude and longitude. The GP of the well-known stars must be known (youtube.com/watch?v=UV1V9-nnaAs)

aw-bib, 2020-12-17:
Maybe  a classic is of help in further research: 

"Celestial Navigation for Yachtsmen / Mary Blewitt", 13th ed., 2017, 78 p., Adlard Coles Nautical,  978-1-4729-4287-6.

It is actually a good read and quite practical (IOW not for the mathematician, who wants all proofs ;). It also gives estimations for the precisions achieved.

Depending on the precision of the measurements taken, however, you might end up quite a bit off. Too much off for hiking, I fear, if you only take the above measurements into account. The _classical method_ relies on a precise time, the use of a sextant to measure the heights, and the nautical  almanac at hand to really _know_ the positions in question and a few tables for corrections. The almanac may be addressed by brute force calculation.

kylecorry31, 2020-12-20:
After some testing, I found that using the altitude of the north star is too imprecise to get the latitude using just a phone. And identifying the precise time of solar noon may be difficult to do.

I may list this in a guide instead, and potentially create a tool around solar noon, as it is more feasible to do than the north star method (using a shadow stick). 

I think the best option for finding location without GPS will be triangulation using existing beacons.

aw-bib, 2021-01-02:
BTW: beyond hiking, but one ingredient for doing the "celestial method" using a sextant, is a well known time. IRL, you take your measurement and once you have the sun down to the horizon you call out "now" and your colleague notes the time up to the second. (Or, presses a stop watch, to "transfer" the time to the ships chronometer.)

Taking the time is almost perfectly addressed in #306: finally, a clock in UT that syncs against GPS and displays seconds. One could now imagine just pressing the clock in TS to jot down the required time. Probably, add the sextant reading, one click more for upper rim/lower rim and another for sun/moon to get a complete log the entry. Potentially a free text to be able to include stars. (And for cross checking add current position from GPS, if available.) You could then take this to the chart table and work out the details.

Of course one could call the issue solved if one has a GPS fix, so no need to bother with a sextant. However, if you're about to learn how to use the sextant, it is a nice idea to avoid an additional error due to a wrong clock reading. `13:24:14 or was it 13:24:41` in the heat of the battle...

closed 2021-01-09 by kylecorry31