kylecorry31/Trail-Sense #338: A display issue

franklinbill, 2020-12-28:
Dear developer!
In the process of using, I found a bug in layout / display. As shown in the figure, in the first page, the icon and the following text overlap each other. This may be an adaptation problem. I suspect it is because **display and font size** are set to large (mainly due to **display size**).
Because of some problems with my eyesight, I often use the largest font and display. I hope that the page can be set to scroll, so that the problem caused by the display size of large can be solved.
[image]
App version is 1.0.1, my mobile phone is Google nexus 6, screen resolution setting is 1920* 1080, **_font size and display size are the largest ! ! !_**, Android system is 8.0.
The above is a description of the problem. The overall experience of the software is good. You can see that the design is very attentive. Thank you very much for the efforts of the developers all the time. I hope to receive a reply.
[image]

kylecorry31, 2020-12-28:
Thank you for the report, I'll adjust this screen accordingly - as well as see if any other screens are affected.

franklinbill, 2020-12-30:
Thanks*100000 for your reply!

commit e4be7867 (2021-02-06): Improve onboarding pages

Closes #338

Closes #227
closed 2021-02-06 by kylecorry31