kylecorry31/Trail-Sense #340: Keep phone unlocked while navigating

kylecorry31, 2020-12-29:
Similar to what Google Maps does when navigating (press power button to resume app).

Leopardus4, 2021-02-06:
I see that this feature is not assigned to 1.7 milestone, so I'm leaving it here only for future reference.

(Disclaimer: I'm not android nor kotlin developer). To enable presence on lockscreen, there is probably something to do with [setShowWhenLocked()](https://developer.android.com/reference/kotlin/android/app/Activity.html#setShowWhenLocked(kotlin.Boolean)) function or [showWhenLocked](https://developer.android.com/reference/kotlin/android/R.attr.html#showWhenLocked:kotlin.Int) value (or maybe [inherit-](https://developer.android.com/reference/kotlin/android/app/Activity.html#setinheritshowwhenlocked) equivalents?).

OsmAnd project implemented this feature, but unfortunately using deprecated FLAG_SHOW_WHEN_LOCKED and FLAG_DISMISS_KEYGUARD flags ([here?](https://github.com/osmandapp/OsmAnd/blob/662959453753f179a15d347fcc4e66233f98642d/OsmAnd/src/net/osmand/plus/activities/MapActivity.java#L1926)).

I found also another open source project using this feature, OpenTracks (unfortunately written in Java). In this project, it's implemented to work with both new and deprecated api, and with nice option to enable/disable it in settings ([this commit](https://github.com/OpenTracksApp/OpenTracks/commit/c8ae3b83b5e9a2503420281019dbf05cdb3b70ef), and small style change [here](https://github.com/OpenTracksApp/OpenTracks/commit/a453c8ec1e5cda467271a316c4d933009cec562a)).

---

OsmAnd has also nice feature, allowing to resume app via gesture, even without the need to touch power button (using proximity sensor) - but it's probably idea for another feature request (not even sure whether this use case is desired/useful).

kylecorry31, 2021-02-06:
Thanks! I'm not sure when I'll get to this, but that will be really useful for when I do

qwerty287, 2021-05-14:
I'd like to work on this, is this ok?
(I would try it with [this](https://github.com/kylecorry31/Trail-Sense/issues/340#issuecomment-774430879) suggestion)

kylecorry31, 2021-05-14:
Sure, my preference would be to have it implemented in a similar way as [this](https://github.com/OpenTracksApp/OpenTracks/commit/c8ae3b83b5e9a2503420281019dbf05cdb3b70ef)

And it should only be enabled when there is a destination beacon or bearing. There should also be an option in the navigation general settings to enable/disable this. 

Thanks!

qwerty287, 2021-05-14:
Isn't [this](https://github.com/kylecorry31/Trail-Sense/issues/340#issuecomment-774430879) and [this](https://github.com/OpenTracksApp/OpenTracks/commit/c8ae3b83b5e9a2503420281019dbf05cdb3b70ef) almost the same? OpenTracks just has a compatibility to older Android version than Oreo (what would be good to have)

qwerty287, 2021-05-14:
But the compatibility for Android Nougat and older is marked as "Deprecated" by Android Studio, should I add it?

kylecorry31, 2021-05-14:
@qwerty287 they are the same, I just wanted to pull the link to the code sample out of the comment above for reference.

And yes, I'm expecting the compatibility version to be deprecated (which is why I like the approach by OpenTracks, pulling it into a method which changes behavior based on Android version) - so fine to add that

PR #677 (2021-05-14): Show TS on lock screen while navigating
Should resolve #340
closed 2021-05-14 by kylecorry31