kylecorry31/Trail-Sense #347: Add tools as home screen widgets

kylecorry31, 2020-12-30:


kylecorry31, 2021-01-02:
(quick actions to the tools)

aw-bib, 2021-01-19:
Random idea: if their content is dynamic they'd also add an informative aspect. E.g. a small barograph leads to weather, moon phase (probably with the next raise/set event) to astronomy.

kylecorry31, 2024-11-10:
Closing this in favor of https://github.com/kylecorry31/Trail-Sense/issues/2415 which contains both in app and homescreen widgets (ended up being really easy to create widgets).

kylecorry31, 2024-11-10:
[image]

michaelblyons, 2024-11-23:
That is many widgets.

kylecorry31, 2024-11-23:
> That is many widgets.

This is only about half of them :)

The rest can be found here: https://github.com/kylecorry31/Trail-Sense/issues/2686

My goal is to let users create a full custom Trail Sense dashboard on their homescreen with the information they care about (rather than me choosing what that information is).

Here's my homescreen dashboard for reference (and I have a couple widgets on the first page of my homescreen - sun and flashlight):
[image]

closed 2024-11-28 by kylecorry31