kylecorry31/Trail-Sense #359: Time is displayed in UTC

dominix, 2021-01-03:
Using AOSP  /e/ time is displayed in UTC and no way to set it to my timezone.
It is a bit counterintuitive to read sunshine at 8pm.

dominix, 2021-01-03:
Same problem on regular Android on Samsung tablet, here we are in GMT-10 (Honolulu/Tahiti)

kylecorry31, 2021-01-03:
Is this in the astronomy tab? I haven't seen this before, can you send me what your system time settings are?

kylecorry31, 2021-01-03:
It is currently configured to use the system default timezone, which is returned by Android itself

dominix, 2021-01-03:
[image]

kylecorry31, 2021-01-03:
Can you double check the location settings in Trail Sense just to confirm they are correct? That's Settings > GPS in TS

kylecorry31, 2021-01-03:
(this looks more like the location is incorrect rather than the times being in UTC)

kylecorry31, 2021-01-03:
Yes I can confirm that this is what happens when the location is 0, 0

[image]

kylecorry31, 2021-01-03:
So that means either the GPS can't get location, or the permission was denied and the location override needs to be set in the GPS settings of TS

dominix, 2021-01-04:
After setting GPS manually,  correct time is displayed. 
[image]

dominix, 2021-01-04:
Verified permission,  they are correct.

kylecorry31, 2021-01-04:
So just to confirm, it doesn't work properly for you when GPS is set to auto, but does when you manually override it? If that is the case, can you re-enable auto GPS and then verify your location appears on the navigation tab?

(if I misunderstood, and you are all set, feel free to close this issue)

kylecorry31, 2021-01-04:
There's also a "Sensor details" option on the settings tab of TS - if you could send a screenshot of that, it would be appreciated to help figure out what's going on - though, it may just be a very delayed GPS reading.

dominix, 2021-01-04:
It definitely require GPS not to be in automatic,  eventually getting data from GPS  works. Will try further when back home.

dominix, 2021-01-04:
Everythings at true, except hygrometer.

kylecorry31, 2021-01-04:
Also, something that I've seen before is the device is set to use the GPS in "battery saving" or "network" mode, but Trail Sense requires that the device is using "device only" or "high accuracy", as it doesn't use network location. Not sure if that is coming into play here though. I suspect it is a slow reading (and I have an open issue to add a timeout there) or that the satellite count isn't supported on your device, which would lead to slow readings.

dominix, 2021-01-04:
Both tested device are using regular gps mode. No battery saving. 
The gps works well when using the button "get location from gps"

kylecorry31, 2021-01-04:
If you click the get location from GPS button, then re-enable auto GPS, and then immediately go back to the astronomy page, does that help at all? I'm going to put in that condition check for when the GPS is 0,0 that I'll push out in the next update - I suspect it will resolve this, as it forces a GPS update in that case.

dominix, 2021-01-04:
Yes it helps.  Time is correctly displayed.

kylecorry31, 2021-01-04:
Thanks! That confirms my theory, which I will provide a fix for in the next release. If this happens again, you should be able to force an update by using the above method.

kylecorry31, 2021-01-14:
This should be fixed in version 1.6.0 (will be releasing over the weekend). I'm going to close this issue, but if after installing 1.6.0 this happens again, feel free to reopen.

closed 2021-01-14 by kylecorry31