kylecorry31/Trail-Sense #369: Add heart rate tool

kylecorry31, 2021-01-08:
With a dedicated sensor or camera.

Could be useful to check for signs of overexertion or dehydration.

kylecorry31, 2021-01-08:
https://www.fruct.org/publications/fruct13/files/Lau.pdf

marcodiegomesquita, 2021-01-18:
Also check https://f-droid.org/en/packages/com.jwetherell.heart_rate_monitor/ . I use it and I think the results are reasonably accurate.

kylecorry31, 2021-01-29:
After some consideration, I will not be adding a heart rate calculator to Trail Sense. Instead, I will be creating a separate Health Sense app: https://github.com/kylecorry31/Health-Sense  

I did this because it will likely have a different userbase than Trail Sense / I don't want Trail Sense to start getting into medical related stuff.

closed 2021-01-29 by kylecorry31