kylecorry31/Trail-Sense #370: Determine current location from dead reckoning

kylecorry31, 2021-01-09:
1. Select beacon (last known location)
2. Enter distance traveled from the beacon (straight line)
3. Mark direction to the beacon
4. Location is calculated

This might not be accurate, as judging distance traveled is difficult. The most accurate measure for this is pace counting with a known pace length.

aw-bib, 2021-01-19:
> The most accurate measure for this is pace counting with a known pace length.

Agree. If this is measured it would be a good start for the calculation.

Average speed and time travelled might also be a reasonably good measure. This is basically the input you'd use for planning the hike, so one may have at least an idea about it. [Naismith's rule](https://en.wikipedia.org/wiki/Naismith%27s_rule) comes to mind: `Time taken = 4km/h + 1min per 10m climbed`. Where the magic `4` is a bit of a personal variable.

(Allowing a base on speed/time would also allow to include other activities then walking.)

aw-bib, 2021-06-21:
isRelatedTo: #142

kylecorry31, 2021-11-14:
Closed for the same reason as #142

closed 2021-11-14 by kylecorry31