kylecorry31/Trail-Sense #391: Add heart rate calculator

kylecorry31, 2021-01-15:
Start a timer and count your heartbeats. When the timer ends, enter the beats to get your BPM.

marcodiegomesquita, 2021-01-18:
I have a more general idea: a tool can be used as counter, chronometer, pulsometer, frequencymeter and tachymeter. The idea is kind of a chronometer with a few more options. I should have 6 buttons and 2 states. The states are PAUSED and RUNNING. The 6 buttons should be:

 - Reset counter
 - Reset chronometer
 - Increment counter
 - Decrement counter

 - On PAUSED state:
   - Continue
   - Continue and increment

 - On Running state:
   - Pause
   - Pause and increment

To accompany the buttons, it should have the following indicators:
 - Elapsed time
 - Events count
 - Events frequency (Hz, per minute, per hour, per day)
 - Average elapsed time between events

The usage very simple:
 - To go to a reset state: pause, reset counter and chronometer.
 - To use it as a counter: reset and simply use the increment counter button.
 - To use as a chronometer: trivial
 - To use it as a pulsometer: reset, start with "continue and increment", increment the counter on each beat and stop with "pause and increment". The longer the sampling, the more precise it is.
 - To use as frequencymeter: same as above
 - To use as tachymeter: reset, start when the event begins, pause it when the event ends and increment the counter.

A very simple but very powerful tool.

kylecorry31, 2021-01-18:
@marcodiegomesquita Thank you for the suggestion. I will consider this but I need to do more research on how useful that would be in a hiking/survival context.

The current tools which use time have specific calculations attached to them (different than the proposed calculations), for example:
- Lightning strike distance uses the speed of sound to calculate the distance between two events
- Cliff height uses a physics equation for uniform acceleration
- Water boil timer is just a timer (count down) that gets populated based on elevation
- (future) Heart rate only makes sense in beats per minute (which this idea would handle, but would also show other frequency values that might not make sense in this context)

marcodiegomesquita, 2021-01-18:
The counter can replace pace count beads. Together with timing, it allows to estimate pace, speed and distance if size of a step is known.

In the dark, you can count the steps in a direction incrementing using the counter. When getting back to where you were, decrease the counter on each step; when it reaches 0 you know your are close to where you were initially. A sound warning or vibration can be added when the counter reaches 0 to help the user.

A tachymeter can be useful as a speed meter if a trail has distance marks or if walked distance can be estimated.

It also allows to estimate how long a process can be repeated per hour. Suppose someone has to prepare a tool for all members of  group; a tachymeter can be used to estimate how many tools will be ready in an hour.

I think these are very simple and useful features and some of them are common on advanced watches.

kylecorry31, 2021-01-18:
Thank you, I will consider it, but will not guarantee I will implement that. I have something else in mind for the pace beads (pedometer)

I also want to do more research into how watches implement that. I'll create a new issue to track that feature.

kylecorry31, 2021-01-29:
After some consideration, I will not be adding a heart rate calculator to Trail Sense. Instead, I will be creating a separate Health Sense app: https://github.com/kylecorry31/Health-Sense  

I did this because it will likely have a different userbase than Trail Sense / I don't want Trail Sense to start getting into medical related stuff.

closed 2021-01-29 by kylecorry31