kylecorry31/Trail-Sense #397: Add daily weather outlook

kylecorry31, 2021-01-16:
A notification every morning that uses longer-term changes in pressure (24 or 48 hours) to determine what the weather will likely be compared to yesterday (ex. worse than yesterday, better than yesterday, about the same as yesterday). Corresponds to the daily weather trend on the weather page (which only appears when a change is expected)

aw-bib, 2021-01-19:
There was an idea on this line in #179 constructing a tangent at a given point to get a clear tendency reading.

Allowing this "tangent" to be constructed between two points that are _t_ (say t=3h 6h 12h 24h 48h) apart might be an interesting idea. Probably, giving a better "feeling" on what's going on than a number.

kylecorry31, 2021-01-19:
So the daily outlook that's currently on the weather tab uses the linear regression (slope) of the 48 hours to generate its prediction, I plan to use something similar here - it is very similar to the tangent idea

commit 9ac79a46 (2021-03-06): Add daily weather notification

Closes #397
closed 2021-03-07 by kylecorry31