kylecorry31/Trail-Sense #402: Weather forecast: unreliable if changing height

jer194, 2021-01-16:
Currently 'weather' is predicting a storm if climbing up a hill (50 m is enough).
Maybe the algorithm could be improved by including height information from GPS
(if available) ...

kylecorry31, 2021-01-16:
Hello,

Yes, that is the setting you are looking for - all the altitude changes currently exist in the released version (in that barometer and thermometer settings), and there are no changes in the upcoming version.

There are only two ways to handle this:
1. Adjust for altitude using GPS (you can do this by toggling off Ignore rapid altitude changes and turn off rapid pressure changes, then set the traveling altitude change threshold to 0)
2. Ignore rapid altitude changes (which are what the traveling altitude/pressure adjustments are for) - this will give better results as the GPS altitude has a lot of noise in it, and you can't really get a good prediction with both variables changing (altitude and pressure)

Unfortunately, I have yet to find a perfect solution to this, but this discussion may help find better values for traveling thresholds (I will have a user guide coming soon): https://github.com/kylecorry31/Trail-Sense/issues/241

jer194, 2021-01-17:
> all the altitude changes currently exist in the released version (in that barometer and thermometer settings), and there are no changes in the upcoming version.

True, I was just comparing on two devices, and the one without pressure sensor didn't show this option :-)
(my fault)

JensKorte, 2021-05-02:
1. <https://en.wikipedia.org/wiki/Differential_GPS> could be a way to make the height calculation better, but it doesn't fit to the "use local sensors" idea.
2. When moving around, often the same places are visited again. How about using the height of the saved locations for correcting the GPS height. An extreme way would be, if TS would build a big map with heights, which gets improved over time and maybe removed, if they are too old. Using the GPS position the map could provide a height. There could be problems, when two nearly the same positions have different height, e.g. at a cliff. Then the map could deal with it in providing an accuracy level (+-10m or 10-50m) or answering with a "height not defined".

kylecorry31, 2021-05-02:
@JensKorte, yeah Differential GPS does not seem like an option as it requires a shortwave radio

As for the height map idea, that does seem promising and I will explore that, maybe related to #173 (it actually seems much more feasible than using DEM).

As for this issue, I'm currently testing a new calibration mode that uses more advanced filtering and signal processing, and so far it appears to be working well for me (it's going to be an experimental option when I release the next build). It also has less confusing settings than the current calibration method.

closed 2021-05-03 by kylecorry31