kylecorry31/Trail-Sense #403: Add 'language' selection in settings / general

jer194, 2021-01-16:
I want to keep my system language in English, however, I want to use Trail Sense in German.
That's not possible now I guess, and could be solved by adding a 'Language' selection in 'General settings'
(similar to 'Theme').

kylecorry31, 2021-01-16:
Thank you for the context behind this - I will consider adding this.

kylecorry31, 2022-05-30:
If it is too complex to support all languages, then add english at first and expand to more languages later. I believe it will be about the same complexity to support all languages though using the new support library.

inkh0rn, 2022-06-13:
Binaryeye done that, please take a look at how it was [implemented](https://github.com/markusfisch/BinaryEye) 

[image]

kylecorry31, 2022-08-21:
https://developer.android.com/guide/topics/resources/app-languages#androidx-impl

Blocked until appcompat 1.6.0 has a stable release: https://developer.android.com/jetpack/androidx/releases/appcompat

kylecorry31, 2022-08-21:
In the meantime, if you have a phone running Android 13 (currently only Google Pixel phones), the next release of Trail Sense (4.4.0) will support the Android system settings app language.