kylecorry31/Trail-Sense #406: Tools / Battery: 'Charging' and 'Discharging' are swapped

jer194, 2021-01-17:
Shouldn't it read 'Discharging' instead of 'Charging', when not connected to a charger?
(and vice versa)

kylecorry31, 2021-01-17:
That's odd; does it show a capacity (mAh) and current draw for you?

jer194, 2021-01-17:
At least on my 'willow' it does; cf. attached graphs.
[image]
[image]

kylecorry31, 2021-01-17:
On the 'willow' device, does the capacity shown increase when connected to the charger (and alternatively, does it decrease when disconnected)?

kylecorry31, 2021-01-17:
I made a few changes to the battery screen in my latest push to main. Based on your post in the New Feature Ideas discussion, it looks like you are building from source. If so, would you be able to try out the latest version on the main branch?

(if not, I can generate a debug build and upload it to dropbox)

jer194, 2021-01-17:
Capacity decreases when disconnected (and vice versa), seems to be working the right way.
I'll try to build with current version, and am going to check, in case I'll succeed.

kylecorry31, 2021-01-17:
Okay, thank you. 

The odd thing is that it works correctly on my device, and according to the Android documentation it is set up properly (positive current means charging, negative discharging - which is what determines the charging message).

jer194, 2021-01-17:
Done:
- Version is now 1.6, so I assume I'm using the current sources.
- Behaviour is still the same as before ..
The value behind the Charging/discharging message (below the bar) is always positive, I assume it 
is not the original value which the decision on charging/discharging is based upon?

kylecorry31, 2021-01-17:
Thank you - and does the arrow next to the percentage change at all?

And no, it is not the original value which the decision for that message is made on, it updates every 20ms with the latest current data from Android's OS (which update frequency there is determined by the manufacturer).

jer194, 2021-01-17:
I seem to have lost the arrow (I remember having it seen in the past, and was wondering why it didn't change, or what it was supposed to indicate).

Another observation: on my S4 the charging/discharging seems to be indicated the right way, though the total charge there is always zero, like on the Moto G4.
I need to go out, maybe I can check te source code myself later (not saying that I'm going to understand anyting 8-) )

kylecorry31, 2021-01-17:
Okay, thank you for testing that. I'm going to make a few more changes and I'll let you know when they are in place. 

The arrow currently uses the change in capacity to determine which way it points - so if the capacity doesn't change, it won't show up (which looking at your screenshots, it doesn't seem to change other than the initial load). If the capacity is reported at 0, it uses a different metric to determine the arrow.

jer194, 2021-01-17:
On the S4, new build, I just saw the arrow again, at it seems to do the right thing (going up when charging eg).
Total charge is gone now (before it was at '0' constantly.

Could it be an option to open a dedicated Telegram group for this app btw?
Thanks  - Jakob.

kylecorry31, 2021-01-17:
Thanks! I'll look into the Telegram group, I've never used it before.

kylecorry31, 2021-01-17:
I think I fixed these issues now - when you get a chance, can you confirm it is behaving properly? (no rush, whenever you have a chance)

For reference:
- The arrow is now determined by the Android OS (no longer tied to the capacity, which is only available on some devices)
- The charging message will match the arrow regardless of if the current was reported as positive or negative (I'm guessing some manufacturers have that opposite)
- If the device does not report current usage, it falls back to AC, DC, or Wireless charging - which is something all devices should report

jer194, 2021-01-17:
Working as expected now, thanks!

'Charging' and 'arrow up', when charging, and vice versa.

Some details:
- Current charge is not displayed on S4 and MG4+, on willow it is there (not changing very often it seems).
- S4 shows always '0 mA' for charging/discharging current, might be a ROM problem.

Thanks for the quick fix!

kylecorry31, 2021-01-17:
Awesome, thank you for testing!

The 0 mA is probably because that property isn't implemented either in the ROM or it lacks the appropriate hardware. I'll put a check in place to fall back to the charging type when 0 mA (or something that round to 0mA) is present.

closed 2021-01-17 by kylecorry31