kylecorry31/Trail-Sense #412: Pressure graph not populating in background

foo-git, 2021-01-22:
**Information**
- OS: LineageOS 17.1/Android 10
- TrailSense v1.5.3
- [X] I read the FAQ and made the app not battery optimized

**Story**
The pressure graph does not automatically populate itself, requiring manual checking/set point of the pressure to populate the graph. App-wise I enabled the following features:
- [X] Weather monitor in background
- [X] Weather notifications
- [ ] Show pressure in notifications
- [X] 15 min refresh interval
- [X] Force weather updates
- [X] Weather updates as foreground activity

As soon as I close the app, the graph is not populated. It should update in background.

I'd be happy to provide more information if needed, as this issue is the only problem I have using your app. Thanks for that!

kylecorry31, 2021-01-22:
Hi, thank you for the detailed report. 

There are a few things you can try here:
1. Uncheck the force weather updates (I find this works better for me on the Pixel 3a, Android 11) - it forces it to use a different task scheduler
2. Under TS Settings > Barometer and thermometer, what happens if you uncheck use sea level and go to the weather tab?

foo-git, 2021-01-22:
> Uncheck the force weather updates (I find this works better for me on the Pixel 3a, Android 11) - it forces it to use a different task scheduler

Thanks! This fixed my issue for now, I will check tomorrow if it logs the pressure also during the night. But I'm confident :)

As a suggestion, would you mind adding 1. to the FAQ? At least for me the "force weather update" button worked somewhat counter-intuitively, as I expected more robust updates with a battery-penalty trade-off.

kylecorry31, 2021-01-22:
Glad to hear! I Will do that, and it is a bit counterintuitive and I may change the wording to legacy scheduler (like I did with the legacy compass). I know that on some devices that does in fact do the right thing, but it's all up to the manufacturer / OS on how that works unfortunately.

foo-git, 2021-01-23:
Just confirming that also during the night the graph was correctly populated. Thanks, I'm gonna close this issue.

closed 2021-01-23 by foo-git