kylecorry31/Trail-Sense #417: Cloud identification

kylecorry31, 2021-01-24:
Knowing the type of clouds can help to predict weather (especially when paired with the barometer). 

Using a ML model trained on several common cloud types, the user should be able to use their camera to get an image and then see the most likely cloud type along with some weather predictions based on that.

kylecorry31, 2021-01-24:
(will need a dataset of cloud images)

aw-bib, 2021-04-21:
There is a pretty nice [FactSheet by MetOffice](https://www.metoffice.gov.uk/binaries/content/assets/metofficegovuk/pdf/research/library-and-archive/library/publications/factsheets/factsheet_1-clouds.pdf)  on the topic. Also there are some usually quite good photos available on their [Clouds page](https://www.metoffice.gov.uk/weather/learn-about/weather/types-of-weather/clouds).

Also [Wikimedia](https://commons.wikimedia.org/w/index.php?search=clouds&title=Special:MediaSearch&go=Go&type=image) comes to mind.

But if this forms already enough input for some ML algorithm...

kylecorry31, 2021-10-02:
This will start off as a simple model:
If the cloud cover is above X% it is a St cloud, else it is a Cu. Once it has that determination done, it can use the color of the cloud to further differentiate Ns from St and Cu from Cb. This will be good enough for an experimental release (and these 4 clouds are the main clouds of concern in regard to the immediate weather forecast). As and Sc will likely be classified as St. Ci and Cc will likely be classified as Cu. Cs is hard to detect on a phone camera, but if it is picked up it will likely be classified as St. 

In the future, I would like to detect all 10 major cloud types using a logistic regression algorithm, but more time needs to be allocated to obtain training data and extracting the features in Kotlin.