kylecorry31/Trail-Sense #422: F-Droid didn't update TS to 1.6.0

Leopardus4, 2021-02-01:
F-Droid didn't update (probably even didn't compile) TS to 1.6. Although they updated app metadata [here](https://gitlab.com/fdroid/fdroiddata/-/blob/master/metadata/com.kylecorry.trail_sense.yml#L173) two weeks ago, last compiled version is probably [1.5.3 (aka 31)](https://f-droid.org/wiki/index.php?title=com.kylecorry.trail_sense/lastbuild&redirect=no).

kylecorry31, 2021-02-01:
Thanks, I opened an issue with F-Droid to see if I can resolve it - I don't see any reason why it wouldn't have gotten built.

Leopardus4, 2021-02-01:
Thank you. For reference, I saw there is also similar issue opened for another app [here](https://gitlab.com/fdroid/fdroiddata/-/issues/2276) - maybe F-Droid contributors will describe here what's going on.

Leopardus4, 2021-02-06:
They rebuilt TS ([build log](https://f-droid.org/wiki/index.php?title=com.kylecorry.trail_sense/lastbuild_32&oldid=286863)), but only didn't publish APK in store. When it happen, I'll close this issue.

I still don't understand what happened at their side, but there were problems with another apps, too. One [disscussion](https://gitlab.com/fdroid/fdroiddata/-/issues/2282) for reference.

Leopardus4, 2021-02-12:
1.6.0 [was published in F-Droid store](https://f-droid.org/packages/com.kylecorry.trail_sense/). Yes, I see 1.7.2 tag in Github :smile:

closed 2021-02-12 by Leopardus4

kylecorry31, 2021-02-13:
Thanks for following up!