kylecorry31/Trail-Sense #425: Invalid location error when converting from UTM

ZweiAmTag, 2021-02-03:
As the title says, conversion from UTM to any other format is not possible.
From decimal to UTM | From UTM to decimal
-----------------------|-----------------------
[image] | [image]

kylecorry31, 2021-02-03:
Hmm, that works fine for me. Could you copy and paste the text you entered into this issue? I'm wondering if there's some sort of encoding issue with the characters.
[image]

kylecorry31, 2021-02-03:
And what version of Trail Sense are you using?

ZweiAmTag, 2021-02-03:
I'm using Version 1.6.0.
This is the input: 32T 0425513E 5302733N
Setting the format to UTM in settings and taping the satellite icon next to the input field , so that it inserts the current location in UTM, gives the same error message.

kylecorry31, 2021-02-03:
Thank you, that is really weird, it works when the device language is English, but not German. I'll investigate further and hopefully provide a fix with the upcoming release.

kylecorry31, 2021-02-03:
Fixed in 1.7.0 (releasing over the weekend most likely)

commit 4a1b936f (2021-02-03): Upgrade TS Core

Closes #425
closed 2021-02-03 by kylecorry31