kylecorry31/Trail-Sense #432: Allow use of fused GPS

kylecorry31, 2021-02-07:
Ex. https://android.googlesource.com/platform/frameworks/base/+/6fa9ad4afcd762aea519ff61811386c23d18ddb2/packages/FusedLocation/src/com/android/location/fused/FusionEngine.java

This may require the course location permission if the user wants to use this instead of the regular GPS.

Not sure if I'll actually implement this though when hiking you typically have a clear view of the sky and not so good cell signal.

Lee-Carre, 2022-01-03:
> Not sure if I'll actually implement this though when hiking you typically have a clear view of the sky and not so good cell signal.

Indeed, or it's not overly difficult to move somewhere which does offer clear-sky reception, when really needed.

From what I read, [Fused Location is intended for inner-city scenarios](http://android-developers.googleblog.com/2020/12/improving-urban-gps-accuracy-for-your.html), where low-accuracy is caused by receiving (only) reflected signals, rather than line-of-sight (the key, there, being *only*; if receiving both the original direct, and a delayed reflection, then some filtering addresses the problem well).

For TS, the show-stopping catch is that the [FLP is available via Play Services only](http://android-developers.googleblog.com/2020/12/improving-urban-gps-accuracy-for-your.html), since it uses a bunch of Google's map data for corrections (notably 3D building models). I recall you mentioning in #972 that TS doesn't (and, I presume, *won't*) rely on Play Services.

See barbeau/gpstest#564.

However, since TS isn't only used while hiking in rural areas (#353), depending on the purpose or intent of this feature, could do a simpler sanity-check if #972 is implemented; query both, and check that the GNSS position (or the entire uncertainty-circle) is within the uncertainty-radius of the approximate (network) location.

kylecorry31, 2022-01-03:
@Lee-Carre thanks - the intent of this issue was to re-implement the Fused Location Provider without Google Play Services (so a custom version, similar to the one linked in the first comment) - mainly for https://github.com/kylecorry31/Trail-Sense/issues/972

I will keep this issue a low priority one, to be implemented once all the main TS features are finished - I can't really imagine a scenario under TS' supported use cases where GPS wouldn't be available in which a cell signal would. The only use case I can think of is when the approximate location permission is the only one that is granted, and therefore TS would need to fallback to network location with the appropriate error banner / notice. 

Side note: When I originally implemented TS, I used the FLP and was getting really inaccurate location readings while hiking - switching to just GPS fixed all of that, so I believe it to be inaccurate network locations. Therefore, at most I would want this to be a fallback for when either a GPS signal isn't available (unlikely) or the fine location permission isn't granted.

kylecorry31, 2023-08-28:
I will not be implementing the fused GPS (with cell / wifi signal)

Rationale:
- Cell signal is poor in a lot of hiking areas, so if even available the accuracy will be pretty bad
- The places where this would be used over GPS are outside the use cases of Trail Sense
- From my understanding, the phone will need to query a remote DB using the cell tower ID / wifi SSID to get its location since this isn't broadcasted directly by the devices. From a privacy perspective, that's not great.
  - An alternative would be to package a cell tower DB with Trail Sense, but it would need to be kept up to date, probably is very large, and as mentioned is much worse than GPS when in the woods.

closed 2023-08-28 by kylecorry31