kylecorry31/Trail-Sense #438: Add tilt, incline

Ygarr, 2021-02-12:
Please add angle level meter, angle of slope

Ygarr, 2021-02-12:
[image]

Ygarr, 2021-02-12:
[image]
[image]

kylecorry31, 2021-02-12:
@Ygarr  Hello, this tool already exists, you are looking for the Inclinometer (under the tools tab), or the bubble level

(inclinometer is designed for slope/height measurement while the bubble level is meant to tell if a surface is flat)

closed 2021-02-12 by kylecorry31