kylecorry31/Trail-Sense #443: Add international whistle codes

kylecorry31, 2021-02-16:
Where are you? : 1
Acknowledged: 1
Come here: 2
Help: 3

qwerty287, 2021-04-25:
Is it okay if I implement this? If yes, I would need some more information, especially how the UI should look.

kylecorry31, 2021-04-25:
Unfortunately that's all the info I have here, I'm not even sure if it should be implemented or should just be a user guide. (This was a community suggestion)

As for UI, I haven't been able to find a nice way of doing this, so I'm not sure - maybe a popup when you long press the "help" button for now? That wouldn't be too discoverable, but then again I'm not sure how useful this feature would be - most people only know the help signal

qwerty287, 2021-04-25:
You could create a new "screen" (fragment) for this and a link to the on the main whistle fragment (instead of the "help" button)

kylecorry31, 2021-04-25:
I'm not sure if I like that approach, I would either prefer a popup where the user selects the whistle (to see what it looks like/how it works, still unsure if I want to have this feature) or not have this feature.

qwerty287, 2021-04-25:
OK, I'll see if this is useful (the UI idea).

qwerty287, 2021-04-26:
Ok I implemented this, see the GIF below. ("Trillerpfeife" is whistle in German, "Hilfe" is help in German) 
[image]

I think, the user experience isn't the best, but it's okay.

kylecorry31, 2021-04-26:
That looks good, as mentioned I don't think most people know those codes so I'm find with the usability there. I'm still debating whether it's worth adding it though. And given that pressing help is designed to repeat over and over (the thought was that you could set your device down while it played - expanded by #561 ) - I would be worried that all of the codes would be interpreted as help anyway. The more I think about it, the more I believe it just belongs in the user guide. Feel free to submit the PR, and if I decide it's worth adding this, then I'll merge it in.

PR #626 (2021-04-26): International whistle signals
Should resolve #443 
The loop works - if you start a signal, it's repeated until stopped. I don't know why you shouldn't add it, if you don't need it, you don't have to use it and it's a little bit a "hidden" feature.
closed 2021-04-26 by kylecorry31