kylecorry31/Trail-Sense #449: Add more sky objects to select for compass

swedneck, 2021-02-26:
E.g. the planets, major stars, maybe arbitrary imported objects somehow?

kylecorry31, 2021-02-26:
That would be cool, but I'm not sure if it would provide any useful information for hiking/survival situations that the compass itself wouldn't already provide (i.e. direction with the North star for example). Can you think of any benefits this could provide?

Regardless, I would be interested in creating a simple "star" map app using some of the technology stack I built with Trail Sense.

swedneck, 2021-02-26:
Mainly thinking that it's good to memorize where stars are in case you ever need to navigate by them, also the fact that hiking is conducive to stargazing. 

A separate app or tool may be better though, perhaps a star chart on the face of a compass.

kylecorry31, 2021-02-27:
Yeah, I think I would prefer a separate app in this case. I don't really the 2D (azimuth) to the star would be all too helpful - a 3D stargazing app would be better. Unless you already know what the star/constellation looks like/how to find it, pointing you in that direction might be confusing as you try to figure out which star TS was referring to, as it could be any in that direction. 

Also, I have all the [icons](https://github.com/kylecorry31/Trail-Sense-Icons/tree/main/Astronomy) already, and the astronomy code I built should support arbitrary celestial objects, so it wouldn't be a huge undertaking to support this in another app, and I have some cool ideas for how to do this in a user-friendly way.

kylecorry31, 2021-02-27:
Ideas:
- Allow user to select a celestial object from a list, and using the camera, compass, and orientation sensor highlight that object in the sky with a circle so the user can more easily find it
- Allow the user to see the path of a celestial object through the sky
- Allow the user to change the current date/time to see where the object will be
- Allow the user to see details about a celestial object, such as rise, set, transit times

kylecorry31, 2021-02-27:
[image]

Felt like working on this a bit today :)

Got all the altitude and azimuth calculations working, now just need to display them as a camera overlay

kylecorry31, 2021-02-27:
(FYI, mapping the planets onto a sphere with the camera at the center is no easy task, putting this on hold for a while)

aw-bib, 2021-03-10:
For mere mortals you may leave out planets beyond Saturn as you'd need a telescope to see them. For Mercury you'll need quite some luck as it's so close to the Sun. (Some famous astronomers of the day never saw it in their lifetime.)

As for celestial objects a suitable list may be given by the Nautical Almanac, e.g. the [Air Almanac 2021](https://www.usno.navy.mil/USNO/library/2021-publications/air-almanac-2021/view). On p.2 a list of bright stars including their positions in 2021 and their magnitude is given. (The lower the mag the better to see. Everything beyond about 4 requires really dark nights without light polution. The limit for the unaided eye is somewhere at 6 mag if it is _really, really_ dark.) This would be a quite _practical_ list, of more or less easily identifiable stars, as it's meant for navigation with a sextant.

BTW: [Planisphere](https://f-droid.org/en/packages/org.tengel.planisphere/) would give a nice star chart. [PlanetDroid](https://play.google.com/store/apps/details?id=com.planetdroid) gives some tools for ephemeris calculation and also offers a star finder using the phones camera.

JensKorte, 2021-03-12:
How about using [Sky Map](https://f-droid.org/de/packages/com.google.android.stardroid/) ?

kylecorry31, 2021-03-16:
I'm not going to be building this - was a fun learning experience though!

closed 2021-03-16 by kylecorry31