kylecorry31/Trail-Sense #456: Compass no longer works once the compass settings have been opened

qwerty287, 2021-03-11:
As the title says - if I open the compass settings once, the compass stop working and shows always 0°.

kylecorry31, 2021-03-11:
@qwerty287 hello, what settings do you have set up on the compass? Also, does force closing and reopening Trail Sense work?

qwerty287, 2021-03-11:
I can't change the compass settings, so I'm using the default settings. 
No, closing or force closing the app doesn't work, I have to reinstall it.

kylecorry31, 2021-03-11:
Does it crash when you try to change the compass settings?

kylecorry31, 2021-03-11:
I have seen Android sometimes have issues with reporting sensor data to apps, in the past when a user restarts their phone it resolves the issue. In that case, it appears to be a system issue and there's nothing Trail Sense can do to resolve it on its own.

qwerty287, 2021-03-11:
No, it doesn't crash. When I go back to the Navigation tab, the compass stopped working.
By the way, it displays 4°, but when I turn off the "Geographic North" feature, it displays 0°.

kylecorry31, 2021-03-11:
If restarting doesn't work, can you try setting to compass smoothing to something around 20 on the compass settings? And possibly toggling on "Use legacy compass"

qwerty287, 2021-03-11:
Oh, I found something working: "Use legacy compass" is on per default, so this doesn't work, but if I turn it of, it's working. If I turn it on again, it doesn't work.

kylecorry31, 2021-03-11:
Okay, good to know - I'm not sure why that was on by default. I'll look into that, the non-legacy compass has better performance and device support. 

My recommendation would be to keep that setting off, which resolves the issue here.

kylecorry31, 2021-03-11:
(I'll add that to the FAQ as well)

closed 2021-03-11 by qwerty287

kylecorry31, 2021-03-11:
I actually found the issue (it was being set to true when you first open the settings page), I'm going to default it to false in the next update (and reset that setting for users - I'll be sure to mention this in the release notes). 

Thanks for catching this!