kylecorry31/Trail-Sense #460: Add anemometer (wind meter)

kylecorry31, 2021-03-12:
Using microphone

https://github.com/kylecorry31/Trail-Sense/issues/59#issuecomment-797536448

This will be enabled with the experimental flag.

marcodiegomesquita, 2021-03-13:
From the linked paper, http://www.acoustics.asn.au/conference_proceedings/ICA2010/cdrom-ICA2010/papers/p41.pdf , we can see that the noise level (Laeq) is a linear function of the logarithm of the wind speed. Since sensitivity and response may vary from one device to another, this feature will need at least 2 calibration points.

Assuming a calibration point has wind speed s1 and noise level l1 and a second calibration point has wind speed s2 and noise level l2, the wind speed s from a sampled noise level l is given by:

s = s2*e^((ln(s1) - ln(s2))*(l - l2)/(l1 - l2))

It is possible to use a linear regression to improve accuracy, but I think 2 calibration points may be good enough for initial tests.

marcodiegomesquita, 2021-07-27:
For the record, the paper says the relation between Laeq (an average sound energy over a period of 1s) and wind speed (in m/s) is given by:

  Laeq = 10 * log(v^c1) + c2

and c1 and c2 depends on the shield used:

  No shield: c1 = 6.59, c2 = 17.6
  65mm shield: c1 = 6.18, c2 = -5.9
  90mm shield: c1 = 6.14, c2 = 7.6
  180mm shield: c1 = 5.63, c2 = -7.9

How to calculate Laeq can be seen in https://github.com/Ifsttar/NoiseCapture

marcodiegomesquita, 2021-07-27:
Just remember that log(v^c1) = c1*log(v). But, as discussed earlier, the table is not needed if a two point calibration is used. We can use these values as default and add a calibration option.

kylecorry31, 2022-01-16:
While this feature would be pretty cool - I don't foresee it having any practical use under TS' use cases. 

Using the microphone would give a (very) rough estimate of the wind speed - this is something the user could figure out themselves by being outside where they would be taking the reading anyway (ex. slight breeze or strong gusts). 

I'll give this a bit more thought in case there are edge cases I didn't consider, but as of now I do not plan on implementing this feature in TS (I might do a separate app for it, because I think it's pretty cool).

closed 2022-01-16 by kylecorry31