kylecorry31/Trail-Sense #466: Add tide clock tool

kylecorry31, 2021-03-15:
I've been doing some research into this, and there are instruments called a tide clock. Essentially it uses the lunar day (24h50m) and you calibrate using a single known high tide (something you could get without internet if near an ocean). This only works in areas where the tides are semidiurnal (https://www.oceanclock.com/en/page/8-how-a-tide-clock-works#:~:text=You%20are%20wondering%20how%20precise,true%20state%20of%20the%20tide%20!). It is not accurate to the minute (probably only to within 30ish minutes), but it gives a good estimate of when the tide will be high or low. 

Due to this, this tool will remain experimental and have appropriate disclaimers around its accuracy. I have to experiment, but it may be possible to increase this accuracy by calculating the lunar day duration instead of using the average. 

Supposedly you only need to calibrate it twice a year - or any time you are moving to a different body of water.

The tool will include both a tide clock and a list of tide times (with changeable dates).

It should also alert the user if the high tide should be recalibrated (ex. if too old, or maybe if they change location)

closed 2021-03-16 by kylecorry31

Lee-Carre, 2022-01-04:
@kylecorry31

Further to our discussion in #1068, I've read quite a few pages of the site you mentioned above

> [How a Tide Clock Works](http://www.oceanclock.com/en/page/8-how-a-tide-clock-works)

As I suspected; [the instructions](https://www.oceanclock.com/en/page/4-how-to-set-your-tide-clock) clearly state that using a high-tide on the same day as a new moon will yield the best results. There's even something of an [explanation](http://www.oceanclock.com/en/page/10-tide-clock-settings) as to why, though is lacking rigorous technical detail. It mentions the term ‘lunar coefficient’ (basically, when Sol & Luna are aligned (as at new moon)).

Either there's some maths I don't understand (highly likely), or the algorithm is designed around being seeded with a HT at new moon.

It all sounds like it's trying to do a crude approximation of M2 & S2 harmonic constituents 😄.

I'm also noticing, today (2022-01-04, ~2 days after new moon), that while the predicted times for high-tide only deviate by a few minutes, the predicted times for low-tide deviate by 30-60 minutes. This isn't a complaint, just an observation.
Though, I suppose high-tide is the one to be cautious about, anyway (especially here; several tidal islets around the coast).

I'll continue to monitor it, over the month.

kylecorry31, 2022-01-04:
I suspect I may want to add an optional low tide reading as well to the tide - as the low tide might not always be halfway between high tides depending on the local geography.