kylecorry31/Trail-Sense #469: Feature request

mahdzu, 2021-03-16:
Amazing app! Thanks for that.

Possible Features: 
1. Panic button, you can send an emergency SMS to some selected contacts including GPS coordinates.
2. Red light (screen) to use in the night.
3. In Astronomy tab, a filter for moon phase and/or date.

I would like to contribute in the Spanish translation as well...

Best regards

kylecorry31, 2021-03-16:
@mahdzu Thanks!

1. Unfortunately, due to the lack of SMS and Contacts permissions Trail Sense can't directly send an SMS - I do have plans to enhance the share menu for your current location though: https://github.com/kylecorry31/Trail-Sense/issues/188

This would be a cool idea for a companion app, which wouldn't be limited by a lack of permissions.

2. For the red light, were you thinking of something like this? https://github.com/kylecorry31/Trail-Sense/issues/210 Or more like use the screen as a red flashlight?

3. Can you elaborate a bit more on what you imagine for this filter? The way I'm reading it is that you want a way to search for example when the next full moon is (which is a cool idea, and I'll create a new issue to log that enhancement)

For translations, you can provide them on Weblate: https://hosted.weblate.org/projects/trail-sense/trail-sense-android/

mahdzu, 2021-03-17:
@kylecorry31 Thanks for your reply.

1. The share menu will be usefull.
2. More like a red flashlight, it helps to adecuate your vision to night vision"mode".  Some headtochs have white and red light.
3. Exactly,  find next full moon... or for an specific date get the moon phase.

Nice, I will see the translations link.

kylecorry31, 2021-03-17:
Thanks! I've created an issue to explore using the screen as a light (for both red and white). 

On the astronomy tab you can change the date using the arrows next to "Today" and that allows you to see the moon phase for a specific date. I just created a new issue for improving that "datepicker" to be a calendar dropdown instead of arrows (the arrows are clunky in my opinion)

I'm going to close this issue in favor of the issues I've added based on your suggestions.

closed 2021-03-17 by kylecorry31