kylecorry31/Trail-Sense #483: Red screen light

kylecorry31, 2021-03-20:
Related to #472

qwerty287, 2021-04-17:
Does "red screen light" only mean to turn the screen red (when using the screen flashlight)?

kylecorry31, 2021-04-17:
Yes, I was thinking there would be a little colored box in the top right corner that if the user clicked it, the screen would become red (and the box would then turn white, so the user could click that and it would turn white again, and repeat)

qwerty287, 2021-04-17:
I think, that's an easy task, so I'll implement it.

kylecorry31, 2021-04-17:
Awesome, thanks! Feel free to do the PR directly to main, I'll include it in the 2.1.0 release (I'm not releasing until probably next weekend, I'll probably be done with the rest of the 2.1.0 features this weekend, but have a bunch of testing to do)

closed 2021-04-17 by kylecorry31