kylecorry31/Trail-Sense #516: Import a printed map

kylecorry31, 2021-04-04:
Select and image of printed map (jpg), then have the user select the location of two beacons/landmarks on the map. This will allow TS to determine the latitude and longitude of all points on the map and see their current location / nearby beacons.

Or with one beacon/location if the scale is provided (ex. Take a picture of a trail sign with the you are here marker)

kylecorry31, 2021-04-04:
Workflow:

User clicks the maps icon on the navigation page / tools. They can select a pre-existing map or add a new one. 

If they select a preexisting map, it loads the image and plots the current location, any beacons on the map, and the destination if available. The user can long-press the map to share a location, create a beacon, or navigate to that point (temp beacon). While navigating, the same navigation panel will be present at the bottom. The user can also drag or zoom in/out of the map. The user can select a new beacon to navigate to by clicking on it on the page and selecting 'Navigate'. Finally, the user can choose a new map to show by either pressing back or by clicking a close icon in the top right. 

If the user chooses to add a new map, they will be asked to give it a name, select an image (map needs to be flat with north facing up). Upon opening the map, they will need to enter the location of two beacons (on a diagonal to each other). They can either select an existing beacon or enter a location manually. Once they do that, they can start using the map.

On the map select screen, users can also delete a map or edit it (name, image, locations).

Leopardus4, 2021-04-06:
Such a good idea!

I just think that it may be hard to take a `flat with north facing up` photo `of a trail sign` or similar outdoor map :wink:.
So the best option of this feature would be to additionally transform a crooked image using 3 or 4 beacons - like `QR scanner` apps. But I guess it will require additional work and image processing library :/

kylecorry31, 2021-04-06:
@Leopardus4  Thanks! So far so good, I have a fully working prototype (releasing in experimental mode). I haven't had an issue getting the picture facing properly as most maps have north at top of the signs and it is fairly easy to get a flat shot of it. As for printed maps, it is a bit more tricky since you can rotate it yourself. I'm going to eventually allow you to rotate the map - and I do have some experience with image processing and transformation.

closed 2021-04-10 by kylecorry31