kylecorry31/Trail-Sense #520: Foreground notifications don't go away instantly

kylecorry31, 2021-04-06:
I'm not sure if this is fixable, I notice this happens with other apps and canceling the notification after the service is stopped doesn't instantly remove it, so it may be something with the OS

https://github.com/kylecorry31/Trail-Sense/issues/518

kylecorry31, 2021-04-12:
This appears to be intended system-level behavior: https://issuetracker.google.com/issues/147792378

https://stackoverflow.com/questions/59526254/foreground-service-notification-always-shown-for-at-least-5-seconds

I don't agree with it, but it's what we got.

"[foreground] notifications must be visible for at least 5 seconds"

I guess that is to prevent apps from quickly doing stuff without the user noticing because they rapidly cancel their notification.

closed 2021-04-12 by kylecorry31

qwerty287, 2021-04-13:
Maybe you could add this to an user guide.

kylecorry31, 2021-04-13:
I can add that to the FAQ, but I don't think it belongs in the user guide