kylecorry31/Trail-Sense #525: Import a map with the camera

kylecorry31, 2021-04-09:


kylecorry31, 2021-04-17:
Taking a photo with the camera requires and external storage permission, and I don't want that (to get the full sized bitmap). Until I find an alternative, this is on hold.

kylecorry31, 2021-04-18:
If the camera features are on, this can be done directly from TS.

Lee-Carre, 2021-12-07:
You might take some inspiration from [MapEver](http://www.github.com/diedricm/MapEver); an older app which tackles the same idea.

kylecorry31, 2021-12-07:
@Lee-Carre that's really cool - I didn't think there were other apps that did this! Looking at the code, I can see that they used a more complex algorithm than I did, it might be worth seeing why / if an approach like that would be beneficial here.

(unfortunately, it doesn't seem to work on Android 12 anymore - I'll have to give it a try on an emulator)

closed 2022-04-14 by kylecorry31