kylecorry31/Trail-Sense #535: Add tracks

kylecorry31, 2021-04-11:
You can navigate with a track on the radar compass or map, and select them from a new tab on the beacon screen. They can only be imported from a GPX file.

Navigation to a track will just show the track and your current position (and the backtrack history)

kylecorry31, 2021-04-11:
Navigating to a track when nearby/radar is disabled will switch to the radar compass

qwerty287, 2021-04-11:
TS should send a notification or inform you if you leave the track.

kylecorry31, 2021-04-17:
[image]

kylecorry31, 2021-11-10:
Closing this as you can display paths on the radar compass and map in 3.0.0

closed 2021-11-10 by kylecorry31