kylecorry31/Trail-Sense #536: Localize cardinal directions

kylecorry31, 2021-04-11:
- Coordinates
- Compass

kylecorry31, 2021-04-18:
#567 should resolve the compass part of this, the coordinates will need more changes in TS core for this to work.

Leopardus4, 2021-04-19:
Localization of TS in general is great, but in this particular case "it's complicated" :wink: 

For example, Polish terms for north and south are "północ" and "południe", so they don't have one-letter abbreviations, but longer ones: "pn/pd" or "płn/płd". In fact, it's popular to simply use English `N`, `S`, `E`, `W` letters there. Especially on compass long ones would look weird.

Idk how it looks in other languages (`help/opinions wanted`). Maybe a new setting might be useful here ("use English/regional abbreviations").

Leopardus4, 2021-04-19:
Edit: it looks like directions are already left in weblate in good forms: [pl](https://hosted.weblate.org/browse/trail-sense/trail-sense-android/pl/?offset=1&q=direction&sort_by=-priority%2Cposition&checksum=) vs [de](https://hosted.weblate.org/browse/trail-sense/trail-sense-android/de/?offset=1&q=direction&sort_by=-priority%2Cposition&checksum=). Title of issue misled me :smile:
So using current translations in compass and coordinates should work well (aka: it won't change anything in Polish :wink:).