kylecorry31/Trail-Sense #539: Open recommended apps if installed

kylecorry31, 2021-04-11:


qwerty287, 2021-05-07:
I would try to implement this, is that okay?
I'd need some more information on the UI, if there should be one button that opens the app if installed or separate buttons for download/open. (I know, it was my idea, I would prefer the first one, but what should be the text on the link? "Open/Download"?)

kylecorry31, 2021-05-07:
I don't think it's possible with the current architecture - the links are being loaded by markdown in the guide tool.

Also, if I'm not mistaken android now requires a package query in the manifest which I'm not sure I want to add without contacting the developers of the recommended apps.

Maybe hold off on this for now? I've also marked issues with help-wanted that are ready for community contribution, as a lot of the untagged issues here have some background information as to why they haven't been implemented/are in the state they are.