kylecorry31/Trail-Sense #549: Add list of tide websites to the tide user guide

kylecorry31, 2021-04-11:


qwerty287, 2021-04-18:
I found two sites that seems to provide worldwide information:
1. https://tides4fishing.com/
2. https://www.tide-forecast.com/

qwerty287, 2021-04-18:
(But the first contains so many information, you have to search fir the tides times on the site, so maybe the second is better)

aw-bib, 2021-04-20:
From https://github.com/kylecorry31/Trail-Sense/issues/103#issue-628558525 here would be some worth considering some official sites:

- [UKHO EasyTide](http://www.ukho.gov.uk/easytide/EasyTide/SelectPort.aspx): worldwide
- [SHOM, France](https://maree.shom.fr/): worldwide
- [NTSLF, UK](https://www.ntslf.org/tides/predictions): UK/Ireland
- [NOAA Tidal predictions, USA](https://tidesandcurrents.noaa.gov/tide_predictions.html): USA
- [BSH, Germany](https://www.bsh.de/DE/DATEN/Gezeiten/gezeiten_node.html): Germany
- [Rijkswaterstaat, Netherlands](https://www.rijkswaterstaat.nl/kaarten/astronomisch-getij.aspx) Netherlands

Usually, those pages offer tidal information for some days.

BTW: What I think would really be a cool idea is, if I could specify the page including all my usual parameters and just fetch it to TrailSense. Say, I want to walk from Cuxhaven to Neuwerk, I'd obviously need the tides for Cuxhaven (and pretty precisely ones, to avoid a meeting with the sea rescue). So [the tides at Cuxhaven](https://tableau.bsh.de/views/Gezeitenvorausberechnung/Gezeitenvorausberechnung_Einzelpegel?Kurzname=Cuxhaven&Bezugsniveau=Seekartennull%20(SKN)&Zeiten=Gesetzliche%20Zeit%20(Sommer:%20MESZ%20%20%20Winter:%20MEZ)&:pixelratio=2&:embed=y&:showAppBanner=false&:display_count=no&:showVizHome=no&:alerts=no&:tabs=no&:showShareOptions=true&:toolbar=yes) come in handy.

Similarly, I want to walk to the Black Cave on Arran I'd most likely need [the tides in Lamlash](http://www.ukho.gov.uk/easytide/EasyTide/ShowPrediction.aspx?PortID=0409&PredictionLength=7) for a few days as they have interesting weather up there and and I can not continue my walk form the case if the tide is in.

I could fetch these data in advance, at home while I do have Internet, but I still don't have a really nice way to get it stored.

qwerty287, 2021-04-20:
> From [#103 (comment)](https://github.com/kylecorry31/Trail-Sense/issues/103#issue-628558525) here would be some worth considering some official sites:
> 
>     * [UKHO EasyTide](http://www.ukho.gov.uk/easytide/EasyTide/SelectPort.aspx): worldwide
> 
>     * [SHOM, France](https://maree.shom.fr/): worldwide
> 
>     * [NTSLF, UK](https://www.ntslf.org/tides/predictions): UK/Ireland
> 
>     * [NOAA Tidal predictions, USA](https://tidesandcurrents.noaa.gov/tide_predictions.html): USA
> 
>     * [BSH, Germany](https://www.bsh.de/DE/DATEN/Gezeiten/gezeiten_node.html): Germany
> 
>     * [Rijkswaterstaat, Netherlands](https://www.rijkswaterstaat.nl/kaarten/astronomisch-getij.aspx) Netherlands
> 
> 
> Usually, those pages offer tidal information for some days.
> 
> BTW: What I think would really be a cool idea is, if I could specify the page including all my usual parameters and just fetch it to TrailSense. Say, I want to walk from Cuxhaven to Neuwerk, I'd obviously need the tides for Cuxhaven (and pretty precisely ones, to avoid a meeting with the sea rescue). So [the tides at Cuxhaven](https://tableau.bsh.de/views/Gezeitenvorausberechnung/Gezeitenvorausberechnung_Einzelpegel?Kurzname=Cuxhaven&Bezugsniveau=Seekartennull%20(SKN)&Zeiten=Gesetzliche%20Zeit%20(Sommer:%20MESZ%20%20%20Winter:%20MEZ)&:pixelratio=2&:embed=y&:showAppBanner=false&:display_count=no&:showVizHome=no&:alerts=no&:tabs=no&:showShareOptions=true&:toolbar=yes) come in handy.
> 
> Similarly, I want to walk to the Black Cave on Arran I'd most likely need [the tides in Lamlash](http://www.ukho.gov.uk/easytide/EasyTide/ShowPrediction.aspx?PortID=0409&PredictionLength=7) for a few days as they have interesting weather up there and and I can not continue my walk form the case if the tide is in.
> 
> I could fetch these data in advance, at home while I do have Internet, but I still don't have a really nice way to get it stored.

An idea would be, that TS suggests you the site for your country (if you list a site for every country).

qwerty287, 2021-04-20:
> BTW: What I think would really be a cool idea is, if I could specify the page including all my usual parameters and just fetch it to TrailSense. Say, I want to walk from Cuxhaven to Neuwerk, I'd [...]

I think, the problem with this could be that you would need a new URL template for every site (if you don't use a worldwide site).

aw-bib, 2021-04-20:
> I think, the problem with this could be that you would need a new URL template for every site (if you don't use a worldwide site).

Sort of. I'd actually not over-engineer this and just give it the bare url. Copy&paste from some browser. Most of them use just some sort of port (number or name) and upon call calc from today onward for _n_ days.

IOW, I'd happily paste this 2km long url (BSH is sort of a worst case, but spot on what you require here) once into some sort of bookmark, that I could just "update" once required and have it for offline reference. Similar to a waypoint that fetches a page. In fact the most common urls I have in some markdown document anyway. It just lacks some `wget`/`curl` that can then display the results properly _offline_.

Could be thought of as quite generic in fact, say if you want to fetch a weather chart or the forecast from the mountain weather service.

kylecorry31, 2021-04-20:
For now, I'm going to go with the simple approach here: just display the links to websites in the user guide with the country as the header and let the user find which link they want to click on.

- Determining which country the user is in is tricky (especially near borders) - and Trail Sense does not have a country bounds reference
- Trail Sense won't be able to download the data directly (no Internet permission)
- The tide tool only supports entry of the high tide and is only designed to give a rough estimate of what the tide is currently - I would recommend a dedicated tide app for exact times. For Trail Sense to be more accurate, either the user would need to enter all the tide times they care about (at that point, it feels more like a note than a calculated value) or enter the complete harmonic data which I'm not planning on implemented / that would be a bad user experience

aw-bib, 2021-04-20:
> Determining which country the user is in is tricky

Perfectly agree. 

Even if you know the service to use getting the nearest standard port is not that easy without users (local) knowledge. You learn to know quite some towns in this business. ;)

This was definitely one of the areas I'd not over-engineer. "Fetch _this_ url" would have been my suggestion. Pretty KISS.

> Trail Sense won't be able to download the data directly (no Internet permission)

Without internet permission it would not work of course. Fine with me. Suggestion retracted, as handy it would be :)

> The tide tool [...] designed to give a rough estimate

Still evaluating it in my (known to be difficult) area.

As far as I understand it, it requires a pretty regular sinusoidal tide (read "more Helgoland than Hamburg"). First checks were not too bad in that case as a rough estimate. I still have to fiddle out if there're general offsets/patterns and how large they are. E.g. it seems that I should include the spring delay (IOW set up using 2 day after full/new moon in my area) and that the date used as reference is not too far away from the date of evaluation. Also springs seem to be closer the mark than neaps.

BTW: one could estimate the height of the tide as well using the rule of 12th. (Don't know the proper English term, if there is any.) Its basically a linear approximation for the sin(). Given the hight of high and low water (from some tide table). Divide the height difference by 12. Then in the first hour you'll get/looose the first 1/12th of the water, in the second hour another 2/12, the next hour 3/12, again 3/12 in the next hour, then 2/12, 1/12 and you are at high/low water. (cf. https://de.wikipedia.org/wiki/Gezeitenrechnung#Zw%C3%B6lftel-Regel in German.) This would probably play well with the tide tool.

> enter the complete harmonic data

As far as I understand it, there is no chance to get it calculated to some precision without the harmonics. (BTW: entering them is only part of the solution. Getting grips on them is already quite challenging as they are not published in most parts of the planet. Some time OpenData will take over though...)

kylecorry31, 2022-01-01:
Related to #1099

aw-bib, 2022-01-01:
Note: It seesm that `UKHO EasyTide` does not offer a world wide scope anymore, but now limits itself to the UK, which is a pitty indeed. :(

Lee-Carre, 2022-01-01:
> seesm that `UKHO EasyTide` does not offer a world wide scope anymore, but now limits itself to the UK

From what I learned in producing #1099, this may be because other sources are available. Notably the WOCE computed harmonic constituents for a whole bunch of measurement station datasets.

Harmonic prediction is what we want, anyway.

Lee-Carre, 2022-01-02:
http://marine.meteoconsult.co.uk/marine-weather/tide-times/brest-4/january-2022

commit c05e928e (2022-06-23): Add links to online tide tables

Closes #549
closed 2022-06-23 by kylecorry31