kylecorry31/Trail-Sense #551: hide settings if required sensors aren't available

qwerty287, 2021-04-11:
There are some settings (or the features) that require a special sensor. One example: the feature "Show linear compass" is shown on my device, but this feature doesn't work. So it would good to hide the setting.

kylecorry31, 2021-04-11:
Hmm, what device are you using? The linear compass should work on any device with an accelerometer (basically all phones).

Are you holding your phone vertically while on the navigation tab?

The only two required sensors in Trail Sense are the magnetometer and accelerometer (because the compass uses both of them), which is why settings don't disable for them. The barometer and pedometer disable their respective settings/tools.

kylecorry31, 2021-04-11:
I'm wondering - so the linear compass uses the gravity sensor (I thought this was available on all devices with an accelerometer), while with the compass itself it uses the gravity and falls back to a custom version on the accelerometer. 

I can make that change for the next release to see if that helps.

qwerty287, 2021-04-12:
> Hmm, what device are you using? The linear compass should work on any device with an accelerometer (basically all phones).
>
> Are you holding your phone vertically while on the navigation tab?
>
> The [...]

I use a HMD Global Nokia 5.3, and it should have both sensors (Trail Senses sensor details confirm this). 
Yes, I hold my phone vertically, but nothing happens.

kylecorry31, 2021-04-12:
Thanks, I'll make some changes to the linear compass to add the same custom gravity sensor fallback that's being used elsewhere.

qwerty287, 2021-05-18:
I don't know why, but this doesn't work for me. Maybe it helps you that the inclinometer doesn't work too, it just says "Hold your phone vertically to measure inclination", even if I hold it vertically. I don't know how that's done, but the bubble level tool works fine. Do I have to toggle on a setting in the system's settings?
My phone has gravity sensor is, and it seems to work (the values are correct, I'd say)

(I wasn't sure if this should be fixed in 2.1.0 and F-Droid updated to 2.2.0 today)

kylecorry31, 2021-05-18:
I just checked

kylecorry31, 2021-05-18:
I just checked and I updated the wrong sensor as a part of this issue... I have the poorly named "DeviceOrientationSensor" and "DeviceOrientation", both of which do different things - I only updated one of them (and couldn't test because my device and emulate both had the gravity sensor). I'll update the core library to account for the other sensor and push that out with the next release.

qwerty287, 2021-06-16:
That's still not working for me. But the inclinometer works after updating to 2.3.0, but the linear compass doesn't appear if I hold my phone correctly.

kylecorry31, 2021-06-16:
I'll add more diagnostics to the sensor details tool to help debug this - oddly they are using the same sensor.

kylecorry31, 2021-06-16:
I just tried on my phone and it isn't working for me anymore. I'll see if I can figure out what changed and put out a hotfix.

commit b8c714af (2021-05-18): Update TS Core

Resolves #680 and resolves #551
commit 00ebd3a2 (2021-06-16): Fix device orientation

Closes #551
closed 2021-06-16 by kylecorry31