kylecorry31/Trail-Sense #562: Translate Readme

another-sapiens, 2021-04-16:
Hi there!
First of all, I love your app.

Second, I was wondering how I could help. The app itself already seems to have been translated, but not the F-Droid description. I've seen some apps do it, do you think it would be possible to translate that one, too?
I think it'd be useful, as people would know the app is in their language.  

One way I've seen it implemented is having an extra folder for each language, with a .txt file for the short description and another for the long description in each of the folders.
For another way, a dev said it could be implemented using Crowdin.

~~Disclaimer: I don't know my stuff, just what I've heard from helping translate a few apps here and there~~

qwerty287, 2021-04-16:
See  #538

another-sapiens, 2021-04-16:
@qwerty287 Oh, didn't see it. Should I close this one?

qwerty287, 2021-04-16:
Yes, it's just a duplicate.

closed 2021-04-16 by kylecorry31