kylecorry31/Trail-Sense #582: Make convert a single tool

kylecorry31, 2021-04-18:
The tool itself will have multiple tabs or a list of units to convert. You can select one and it will open the convert screen for that unit.

This will make it easier to add more convert tools in the future (like temperature or pressure)

qwerty287, 2021-05-07:
Hi,
I'm working on this, but I'm not sure if I understood your UI idea. I captured two screenshots, see below. By the way, in my AVD, the distance converter has a wrong format (see the second screenshot), this even happens if you enter something. Is this a bug or another problem?

[image]
[image]

There are still some things that aren't done, I think, the titles in the tab bar should only contain "Coordinates" or "Distance".

kylecorry31, 2021-05-07:
Yeah, that's what I'm imagining, thanks! And as for the format, that's a work in progress (I just haven't created a new release of TS Core that has the updated formatting yet)

And yes the tabs should only contain coordinates and distance (and also the title under the tabs can go away)

PR #651 (2021-05-07): Single convert tool 
Should resolve #582 

Sorry for the commit messages, I had some trouble with my Git installation.

The two first commits should be one, the third and forth should be one too and the fifth removes the title TextViews and the strings that were used there.
closed 2021-05-07 by kylecorry31