kylecorry31/Trail-Sense #590: Toggle sunrise/sunset theme on resume

kylecorry31, 2021-04-19:
Right now it works on create, but it might be useful if TS was in the background and reopened later in the day. This should only be done if it doesn't affect performance / is buggy or is annoying to users. 

Only when the sunrise/sunset theme is enabled.

qwerty287, 2021-04-19:
I tried to implement this, but I think (in the way I implemented it) it could be annoying. See the GIF below.
[image]

kylecorry31, 2021-04-19:
Yeah, that wouldn't be a fun user experience

kylecorry31, 2021-04-19:
Yeah, I think the only way is to recreate the activity, which isn't always desirable. Once Jetpack Compose has a stable release, I'm going to switch over to it, at which point this should be feasible.

kylecorry31, 2021-08-29:
I'm going to close this in favor of using the system options for day/night theme. I don't think I'm going to switch Trail Sense to Jetpack Compose - too much work for that.

closed 2021-08-29 by kylecorry31