kylecorry31/Trail-Sense #591: Incorrect sunrise and sunset times

another-sapiens, 2021-04-19:
Hi there!
I'm not sure if it's just an issue on my end, but the sunset notifications alert that the sun will set at about 15:00, despite the correct time being 3hs after that. Same with sunrises: Today, for example, it tells me that it will be at 2:55 in the morning. 

Have I overlooked something?

kylecorry31, 2021-04-19:
Hello, please see the "sunrise/set times are inaccurate" section of the FAQ: https://github.com/kylecorry31/Trail-Sense#faq

If that does not resolve the issue, can you verify that you are running the latest version (2.0.0)

kylecorry31, 2021-04-20:
I'm also going to make an update (in version 2.1.0) to retrieve the location if possible (requires background location to be granted) before sending a sunset alert. 

I do believe that you will be able to resolve this before that update by ensuring your location is properly set in Trail Sense - if you have GPS disabled (or the location permission denied), you will need to manually configure your location under Trail Sense's GPS settings (it defaults to 0, 0 - on the astronomy page you will see an error banner for this condition). If your GPS is enabled and the permission is granted, you may need to turn off the "require satellites" setting or wait for the 10 second timeout with a clear view of the sky so Trail Sense can cache a GPS reading. The cached value can be verified on the "sensor details" page - bottom of settings.

another-sapiens, 2021-04-20:
I just found out that this may be a problem with my device's settings, but I may have not noticed because it's the only app for which I've enabled the location.

Thank you for your attention, and for your amazing app. Keep up the good work!

closed 2021-04-20 by another-sapiens

another-sapiens, 2021-04-20:
I've manually selected my location and now it works like a charm.

kylecorry31, 2021-04-20:
Glad to hear!