kylecorry31/Trail-Sense #603: Allow refreshing clock signal

kylecorry31, 2021-04-23:
Add a button to refresh the clock - if it isn't too hard also support swipe down to refresh.

qwerty287, 2021-04-23:
By the way, I could do this.

kylecorry31, 2021-04-23:
My preferred approach here would be to get the refresh button in place first, then as a separate PR to get the swipe to refresh if possible (it looks like the docs are targeted toward listview and gridview, which won't be used on the clock tool)

qwerty287, 2021-04-23:
Ok, I had some problems with the update() function, the GPS start function continues directly, so update() is called immediately. Is update() called always in the background?

kylecorry31, 2021-04-23:
Update get's called every 20ms by the invervalometer (private val timer = Intervalometer { update() }, and timer.inverval(20))

The GPS is only queried once - returning false from the method in the start function cancels the subscription - the function parameter in the gps.start(...) call is called asynchronously when a valid GPS reading is received.

closed 2021-04-27 by kylecorry31

Lee-Carre, 2022-01-02:
@kylecorry31

> refresh the clock

Could you clarify what exactly you mean by refreshing the clock;
* the display of the UI?
* Re-getting the current correct time?
* re-triggering the `update system time` function?
* something else?

I ask in order to know if this issue is relevant to some ideas (I'm on a roll, today).

kylecorry31, 2022-01-02:
This feature was the refresh button at the top of the clock tool - it essentially just grabs another GNSS reading as the true time if the previous one timed out or something

Lee-Carre, 2022-01-04:
> grabs another GNSS reading as the true time if the previous one timed out or something

👍
Different to what I was thinking.

Does the clock stay in sync when the GNSS receiver is kept enabled? If not, could it be made to?

kylecorry31, 2022-01-04:
The clock updates its time based on the first GNSS timestamp when you open the clock tool, and only rescans when you click the refresh button - is there a benefit in constantly updating off of the GNSS signal?