kylecorry31/Trail-Sense #635: Determine stride length from GPS

kylecorry31, 2021-05-01:
1. User says they want to calculate stride length
2. Prompt to start walking
3. GPS tracks distance until an accurate value is determined for stride length or the user stops it

Lee-Carre, 2022-01-26:
> GPS tracks distance until an accurate value is determined for stride length or the user stops it

What about continual auto-calibration in the background (much like for barometer-based altitude from GNSS)?

* Stride-length may change over the journey (depending on terrain, conditions, footwear).
* Adapts to whoever happens to be carrying the device (imagine a communal backpack being passed around, to share the load, during a trek).

In such a case, it may be helpful to have a method for triggering a calibration reset.

kylecorry31, 2022-01-26:
@Lee-Carre good idea - I created a new issue for that https://github.com/kylecorry31/Trail-Sense/issues/1164

closed 2022-02-05 by kylecorry31