kylecorry31/Trail-Sense #642: Pressure is confusing between station pressure, barometric pressure and altimeter pressure

gdt, 2021-05-01:
The phone sensor is measuring station presssure, in weather terms.    It might be nice to allow a calibration of that to be entered; I can see that mine is off about 1 hPa from a weather station and a BME280.  That said, I'm still not sure which to believe.

From the configuration screen, I see "Sea Level Pressure".   That seems obviously intended to take some GPS-derived orthometric height and then to try to reduce pressure to sea level.   I would hope that there is some mapping from GPS-derived height and pressure over some medium time scale as GPS height is poor quality.

Change the pressure label to (S), (B) or (A) after the units, or something, so the user can tell which pressure is displayed.  (While there, set the format to %.1f for hPa so the tenth is always shown, not disappearing if 0.)   Maybe people don't want this, and it shouldn't be present.  In that case, in the altitude settings screen, show "(station/barometric/altimeter) pressure" instead of pressure, so that the choice is more in-your-face, and then it probably can be omitted on the main screen.

The notion of barometric pressure, AIUI requires history of temp and humidity.    Phones usually don't have humidity, and most phone temp sensors are battery or bogus.   So I would guess that the pressure shown for 'sea level' is altimeter pressure, reduced for elevation but not adjusted for temp/humidity.  That's fine if so, but it would be nice to be clear, perhaps a checkbox for Station/Barometric/Altimeter, with Barometric greyed out if temp/humidty data isn't available.

(I realize this is an exercise in trying to get decent data out of something that really wasn't designed for it.)

kylecorry31, 2021-05-01:
It appears there are two calibrations here: the station pressure calibration (i.e. does the phone read pressure high or low) and the sea level calibration (using the altimeter). 

Trail Sense does not currently support the first calibration (station pressure), and as for the second - yes it does use a filter applied to historical GPS altitude data (fused with historical pressure data). 

Trail Sense does not distinguish between the mean sea level and altimeter setting pressures (https://www.weather.gov/bou/pressure_definitions), it uses the altimeter reported altitude (as configured in the altimeter calibration settings - this could be considered the Altimeter Setting), with the option of using the temperature sensor (either a dedicated temperature sensor or the calibrated battery temperature sensor, which reads high most of the time - this could be considered the Mean Sea Level setting). 

As you mentioned, a phone's barometer isn't designed with this in mind; it is used to measure changes in floors. I don't think I'm going to adjust the types of pressure shown in Trail Sense though, and here's why:
- The main purpose of the weather tool is to analyze trends in pressure and provide possible insight into what it means (worsening, improving, etc) - the algorithm does not need to know the exact pressure, just how it changes over time (and whether it is high or low for the icon shown, which has decent wiggle room)
- The sea level pressure feature is really designed to remove the effect of changing altitude for the weather analysis algorithm and from my experience comes pretty close to matching the reported MSL pressure from nearby stations in my area

Maybe the real thing that is missing here is a message (on the barometer calibration page) stating that it is expected that the pressure shown in TS may differ from other sources with a link to the user guide on Weather Predication for understanding how TS uses pressure to predict weather changes.

gdt, 2021-05-01:
Certainly fair enough on what you are trying to do.   Where I'm coming from is that people who understand weather know that barometric pressure and altimeter pressure are not quite the same. 

So if you are really calculating altimeter pressure, and the label *in the settings page only* flips from "Station Pressure" to "Altimeter Pressure (sea level)" or something, when "sea level pressure" is on, then I think those people will understand, and others won't really pay attention.

if you're calculating something that's sort of a hybrid between altimeter pressure and barometric pressure, then I guess I'd suggest not to do that.   The world doesn't need another definition.

Agreed that altimeter pressure rise/fall (or barometric if available) is what should be paid attention to.   Or really pressure at any given elevation.

But all of this is not a big deal, and I realize I'm a bit out there -- so if you just want to hit close that's of course your call.

kylecorry31, 2021-05-01:
Yeah, it is definitely a hybrid between the two so I feel that changing the label would add more confusion. I am going to close this and will expand the user guide for weather prediction to include some of the points brought up here.

closed 2021-05-01 by kylecorry31