kylecorry31/Trail-Sense #645: Research raw GNSS measurements

kylecorry31, 2021-05-02:
I'm not really expecting this to go anywhere, but I'm interested to see what's available and if that could be used in any way.

JensKorte, 2021-06-16:
https://en.wikipedia.org/wiki/GNSS_reflectometry
I am unsure if this is of any use. In the [german WP article](https://de.wikipedia.org/wiki/GNSS-R) it sounds like: if a satellite is just above the horizon, it can be used by a GPS receiver to measure the reflections on the surface of the water and tell something about the waves.

kylecorry31, 2021-07-21:
Unfortunately, it doesn't look like there is anything I can get out of the GNSS API that would be useful which the location API doesn't provide.

closed 2021-07-21 by kylecorry31