kylecorry31/Trail-Sense #658: Incompatibility with system torch/flashlight

fekir, 2021-05-10:
Hello,

when Trail Sense is running (ie has been started at least once), I'm unable to turn the flashlight on from the tile.

See attached video.

https://user-images.githubusercontent.com/11912465/117646261-d8c08c00-b18b-11eb-9816-ccf273e768fb.mp4

This is LineageOS 17.1 (thus based on Android 10) on a sony xperia z2 compact (xz2c)

When trying to start the flashlight from the tile, if Trail Sense has been started, I get a notification, but the flashlight is turned off immediately.

From the app, the flashlight works correctly (and can be turned off from the tile), but using the tile for turning it on is much more practical.

kylecorry31, 2021-05-10:
Hi, unfortunately, I can't reproduce this on my device - but looking at the provided video, it may be a version of the device configuration issue mentioned in the FAQ called "I can't turn off the flashlight/white noise or they won't turn on". Can you try that workaround and see if it works? (https://github.com/kylecorry31/Trail-Sense#faq)

fekir, 2021-05-10:
I'm not sure I understand the FAQ, but trail-sense has all permission, even those it never asked on my device, same holds for notifications.

This are the latest entries in logcat when trail sense is running and I tip on the flashligt from the tile

```
05-10 15:45:38.588  1468  1691 E LightsService: Light requested not available on this device. 2
05-10 15:45:38.754   928   928 D libcameralight: msm_flash_v2.cpp chip_device_get_errorsituation: start
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp read </sys/class/leds/led:switch_0//fault_status>[0]
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp chip_vr_torch_turnon: start
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp chip_torch_set_current: current 25000
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp calc_led_resolution_ua: current_uA[13000] led_resolution[0] = 12500
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp current_to_regvalue: current_uA[25000] gIresUa[12500] reg_uA[13000]
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp chip_torch_turnon: start
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:torch_0/ires_ua>[12500]
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:torch_0/brightness>[13]
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:torch_1//ires_ua>[12500]
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:torch_1//brightness>[13]
05-10 15:45:38.755   928   928 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:switch_0//brightness>[1]
05-10 15:45:38.756  1110 11187 I CameraProviderManager: Camera device device@3.4/internal/0 torch status is now AVAILABLE_ON
05-10 15:45:38.756  1110 11187 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=2
05-10 15:45:38.758  1110 11187 I CameraService: Torch for camera id 0 turned on for client PID 2082
05-10 15:45:38.768  1468  1468 V SettingsProvider: Notifying for 0: content://settings/secure/flashlight_enabled
05-10 15:45:38.775  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.779  1110 11187 I chatty  : uid=1047(cameraserver) Binder:1110_5 identical 4 lines
05-10 15:45:38.780  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.785   928   928 D libcameralight: msm_flash_v2.cpp chip_vr_torch_turnoff: start
05-10 15:45:38.785   928   928 D libcameralight: msm_flash_v2.cpp chip_torch_turnoff: start
05-10 15:45:38.785   928   928 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:torch_0/brightness>[0]
05-10 15:45:38.785   928   928 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:torch_1//brightness>[0]
05-10 15:45:38.785   928   928 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:switch_0//brightness>[0]
05-10 15:45:38.785  1110 11187 I CameraProviderManager: Camera device device@3.4/internal/0 torch status is now AVAILABLE_OFF
05-10 15:45:38.785  1110 11187 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=1
05-10 15:45:38.787  1110 11187 I CameraService: Torch for camera id 0 turned on for client PID 9653
05-10 15:45:38.799  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.802  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.806  1110 11187 I CameraProviderManager: Camera device device@3.4/internal/0 torch status is now AVAILABLE_OFF
05-10 15:45:38.806  1110 11187 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=1
05-10 15:45:38.806  1110 11187 I CameraService: Torch for camera id 0 turned off for client PID 9653
05-10 15:45:38.819  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.820  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.822  1110 11187 I CameraService: Torch for camera id 0 turned off for client PID 9653
05-10 15:45:38.846  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.851  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.856  1110 11187 I CameraService: Torch for camera id 0 turned off for client PID 9653
05-10 15:45:38.919  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.924  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.930  1110 11187 I CameraService: Torch for camera id 0 turned off for client PID 9653
05-10 15:45:38.989  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:38.996  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
05-10 15:45:39.004  1110 11187 I CameraService: Torch for camera id 0 turned off for client PID 9653
05-10 15:45:39.067  1110 11187 W ServiceManager: Permission failure: android.permission.CAMERA from uid=10174 pid=9653
[... cut ...]
05-10 15:46:21.917   928  4282 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:torch_1//brightness>[0]
05-10 15:46:21.917   928  4282 D libcameralight: msm_flash_v2.cpp write </sys/class/leds/led:switch_0//brightness>[0]
05-10 15:46:21.917  1110 11187 I CameraProviderManager: Camera device device@3.4/internal/0 torch status is now AVAILABLE_OFF
05-10 15:46:21.917  1110 11187 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=1
05-10 15:46:21.920  1110 11187 I CameraService: Torch for camera id 0 turned off for client PID 2082
```

kylecorry31, 2021-05-10:
So it isn't a permission, it is a notification setting in the Android framework: https://support.google.com/android/answer/9079661?hl=en#zippy=%2Cturn-notifications-on-or-off-for-certain-apps

The setting you use to block an app from sending notifications (when the flashlight turns on in Trail Sense, you should see a notification - if you don't then the notifications are likely turned off for Trail Sense)

kylecorry31, 2021-05-10:
As of right now there isn't a way to disable this - I'll add a setting in the next release. In the meantime if you swap out the flashlight quick action in Trail Sense (Trail Sense settings > Navigation > Right quick action) and then force stop Trail Sense it will should work again (until you open the flashlight tool).

commit 53fc53ac (2021-05-12): Add setting for flashlight system integration

Closes #658
closed 2021-05-12 by kylecorry31