kylecorry31/Trail-Sense #676: Experimental barometer calibration feedback

kylecorry31, 2021-05-14:


aw-bib, 2021-06-01:
Some minor comments on UI:

- It took me quite a bit to find these settings, as all settings get _hidden_ (not only disabled) now as soon as I switch off the GPS receiver. I understand that this _is_ intended, as no correction against GPS if no GPS, but I wonder if they shouldn't just greyed out to know there _is_ something, and/or probably some hint "you need gps to see calibration settings".
- Sliders vs. the graph: the graph indeed seems helpful to find settings, however, it should stay visible once I move down to the sliders. Right now it is scrolled off the top of the page.
- First glance at the graph looks promising. I'll give it a test in the wild.

kylecorry31, 2021-06-02:
- Thanks, I'll improve the GPS-barometer interaction in settings
- I'll see what I can do about that / what works well from a UX perspective

aw-bib, 2021-06-03:
Preliminary results from yesterdays trip looked promising, but it needs longer observation periods, especially as we have a quite stable high at the moment.

aw-bib, 2021-06-10:
Did some more tests on my usual walk (the one that usually triggered a storm warning due to the spike when I descend down to the river). Indeed the algorithm seems to have improved quite a bit. :)

Right now we have a completely calm H in my area so the official curves of barometric pressure are basically a horizontal, flat line. Now I did my usual tour and this is the situation the new algorithm:

[image]

Some minor kinks, but well, now spikes and no (wrong) warnings issued. I just observe a small (constant) offset compared to the official sea level readings. AFAIS this just stems from the simple fact that TS overestimates the "height". IOW if I calibrate against a barometer reading, the calculated height is some 25m lower than the one reported by TS via GPS and set to automatic calibration.

JensKorte, 2021-06-11:
I use the new experimental calibration and I like it. In past I got once in a while a wrong storm warning when I was on the road. Now I don't get any warnings any more because of moving. The last warning was with a falling pressure without a storm. The bad weather didn't come to us, but the clouds told that there is a kind of energy in the air.

I use: (translating the UI back from german) distance to height changes (?) +-19m; height leveling 5%; pressure leveling 50%

kylecorry31, 2021-06-12:
Awesome, glad to hear! Thank you both for the feedback

closed 2021-11-14 by kylecorry31